black-parrot/black-parrot #1020: bp_me_axil_master issues misaligned memory access

Yuan-Mao, 2022-02-02:
## Branch / Commit Hash
top_dev: af8beff

## Area
bp_me_axil_master module
https://github.com/black-parrot/black-parrot/blob/top_dev/bp_me/src/v/network/bp_me_axil_master.sv#L113

## Current behavior
Instead of `m_axil_araddr_o  = io_cmd_header_cast_i.addr[0+:axil_addr_width_p];`,
we might want something like `m_axil_araddr_o  = {io_cmd_header_cast_i.addr[2+:axil_addr_width_p], 2'b00};`(assuming bus width == 32),
so that we issue aligned memory access in AXIL. After getting the 4-byte read data, bsg_bus_pack will then select the appropriate subset of data and send it back to bp I/O cmd.

If this is correct then I can make a PR for this.

dpetrisko, 2022-02-02:
Are you trying to access unaligned registers on the AXI bus? It seems unlikely that software would require this

Yuan-Mao, 2022-02-02:
For example if BP wants to read a 16-bit data from address 0x1002. This BP i/o cmd will translate into a 32-bit read at 0x1002 on AXIL bus in our current implementation.

dpetrisko, 2022-02-02:
Yes I understand. Is there a device that you’re using that has 16-bit registers?

Yuan-Mao, 2022-02-02:
https://elixir.bootlin.com/linux/v5.15/source/drivers/net/ethernet/litex/litex_liteeth.c#L78
I guess when Linux wants to read out the Ethernet packet from IO, it might emit that kind of memory access. Say the incoming packet size is 0x79, the kernel might eventually have to do a 1-byte read from 0x79 instead of doing 4-byte accesses throughout the process.

dpetrisko, 2022-02-02:
The data buffer isn’t a fifo in the controller? We’re reading from a block of memory?

Yuan-Mao, 2022-02-02:
Yeah we are reading from a block of memory. This is compatible with Litex Ethernet core. It is not like, we read data from a fixed 32-bit memory address (if this is what you mean about the FIFO)

dpetrisko, 2022-02-02:
DIscussed offline, but for posterity: the axi spec dictates that clients should ignore the bottom bits such that all requests return data aligned to the bus width. Therefore there is no need to truncate the addresses for reads.

However, writes require shifting the strobes appropriately. I'll leave the issue open for someone to add this feature if they need it

dpetrisko, 2022-07-30:
Closing issue as axi modules have moved to subsystems

closed 2022-07-30 by dpetrisko