black-parrot/black-parrot #1038: Update references to the ISA manual to a ratified version

omasanori, 2022-02-17:
## Branch / Commit Hash
N/A

## Area
docs, perhaps fe and be too

## Current behavior
Currently [the documentation refers to the ISA manual, Volume 1 version 2.0](https://github.com/black-parrot/black-parrot/blob/414747b05833f429c418af6502f13a327ef4df67/docs/testbench_guide.md?plain=1#L28) but the manual has been revised and ratified by the RISC-V International several times. The behavior should not be changed, but it would be better to review the documentation and implementation.

## Expected behavior
N/A

## Proposed solution
Check if the documentation and implementation diverge from the current ratified specifications.

## Analysis
It will probably not affect the functionalities.

## Verification
Review manually.

## Additional Context
N/A

dpetrisko, 2022-02-17:
Hi, thanks for the issue. As you say, it's basically a matter of meticulously checking the differences between 2.0 and 2.1. I believe the most obvious one is that we are now RV64IMAFD_Zifencei_Zicsr since that's been removed from the base ISA

omasanori, 2022-02-17:
Agreed. RVF and RVD extensions may need some attention, as the revision of IEEE 754 referred by the specification was updated, NaN-boxing was specified (I saw this one seems addressed in BSG's HardFloat repository), and such. The memory model might be the most challenging part for reviewing IMHO.