black-parrot/black-parrot #1056: Investigate Zachary Snow's SystemVerilog to Verilog HDL converter (sv2v)

omasanori, 2022-03-04:
## Branch / Commit Hash
`dev`

## Area
tools

## Current behavior
To work around tools with insufficient SystemVerilog support, a SystemVerilog-to-Verilog HDL converter is desired. BSG currently maintains [bsg_sv2v](https://github.com/bespoke-silicon-group/bsg_sv2v) that depends on [Synopsys Design Compiler](https://www.synopsys.com/implementation-and-signoff/rtl-synthesis-test/dc-ultra.html). While it works reliably, getting a license of the proprietary CAD suite is sometimes unreasonable to individual developers and small institutes.

## Desired behavior
Use a free/libre and open-source conversion tool.

## Proposed solution

Check if Zachary Snow's [sv2v](https://github.com/zachjs/sv2v) is sufficient for BlackParrot.

## Analysis

No impact on functionalities. It helps to port BlackParrot by individuals and make BlackParrot more accessible.

## Verification

Convert the code with `sv2v` followed by regular verification.

## Additional context

According to recent [results of SV-Tests](https://chipsalliance.github.io/sv-tests-results/), Zachary Snow's `sv2v` succeeds in processing virtually all of BaseJump STL tests (232/233) and all components of BlackParrot (7/7).

dpetrisko, 2022-03-04:
#171 #604 #682 

indeed we’ve gone down this road. At the time we discovered several bugs in both the tool and BP’s ieee compliance. In the end, we almost got it but couldn’t quite finish. Would accept a PR with support, but the caution is that even though code may parse 100%, it’s not guaranteed to be a bug-free conversion. It may not run code properly (although it should be approximate for running through a synthesis flow)

omasanori, 2022-03-04:
I was unaware of the previous attempts. Thank you for pointing it out. Then shall we leave this issue until a stable conversion is observed and support is implemented?