black-parrot/black-parrot #1082: Blackparrot on doubletrouble

zhongpanwu, 2022-04-03:
Hi,

I was wondering if it is possible to run black parrot on doubletrouble system?

Yunus

dpetrisko, 2022-04-14:
Hi, this repo contains RTL which is board-agnostic. We have taped out BlackParrot several times for systems like double-trouble (using bsg_channel_tunnel for I/O and memory). See [here](https://github.com/black-parrot-examples/bsg_tapeout_v0_gf12/blob/master/toplevels/bsg_ac_black_parrot_quad_core_rc0/v/bsg_chip.v) for an example of a similar setup. This is a very old version of BlackParrot, so it will need to be adapted for minor changes in the toplevel interface.

Can you share more details of what you're trying to accomplish?

black-parrot, 2022-04-14:
Are you guys using Synplify for SystemVerilog synthesis?

yunusdawji, 2022-04-16:
Hi,

Thanks Dan, I will check it out. And yes we are using Synplify for systemverilog synthesis.

Yunus

black-parrot, 2022-04-17:
Hi Yunus,

Then I think you can definitely do it. The Spartan-150’s are pretty big
chips.

The main thing is the I/O system. You would connect it to a host FPGA board
for DRAM and memory traffic. If you use a Zynq host, Dan can probably
suggest some sample code.

M

dpetrisko, 2023-07-18:
Closing for staleness. Let me know if you have any additional questions (or had success!)

closed 2023-07-18 by dpetrisko