black-parrot/black-parrot #1132: Is it possible to use fewer than 64 sets in the instruction cache?

flaviens, 2022-10-25:
Hi there!

## Branch / Commit Hash
Master

## Area
[Instruction cache.](https://github.com/black-parrot/black-parrot/blob/6ce6523d6b3ab91cdb355d53c419f1b9ee1e1e81/bp_fe/src/v/bp_fe_icache.sv#L680-L682)

## Question
Is it possible to have fewer than 64 sets in the instruction cache or will it not work?
I tried to configure the number of sets to a smaller number using packages but received this error.

Thanks!

flaviens, 2022-10-26:
Hi Dan, thank you for your fast response! Interesting stuff.

dpetrisko, 2022-11-14:
what are all of the icache_* parameters you are using in your configuration? It seems you are trying to use a configuration for which the fill_width < bank_width which is currently illegal (each memory access should fetch at least 1 bank)

flaviens, 2022-12-16:
Thank you very much for your response!

closed 2022-12-16 by flaviens