black-parrot/black-parrot #1139: Question: how to pass the first cycles

flaviens, 2022-11-11:
Hi!

My apologies if this information has been asked already or if it stands somewhere underlined already.
I observed that BlackParrot only starts fetching after approx 512 cycles after reset.

Is there a way to reduce these 512 cycles? Minor hacky modifications in the design are perfectly fine if necessary.
The goal is to run repeated short simulations, as in this scenario these lead cycles are lowering performance significantly.

Thanks!
Flavien

dpetrisko, 2022-11-11:
Hi Flavien, the first part of the execution is loading the program into DRAM. This does take a while, especially for larger programs. We provide a backdoor preload flag you can use by setting PRELOAD_MEM_P=1 during your simulation build command. However, it does not appear to currently work with Verilator:

https://github.com/black-parrot/black-parrot/blob/6ce6523d6b3ab91cdb355d53c419f1b9ee1e1e81/bp_me/test/common/bp_nonsynth_dram.sv#L195 

The second thing we do is start in uncached mode and execute from bootrom. If you're trying to reduce the overhead of starting, I would recommend starting directly from the program start address in DRAM. I believe the extra flags would be: 
NBF_CONFIG_P=1 (which will set configuration registers through the TB instead of the bootrom) and adding 
--boot_pc=0x80000000 to the NBF_INPUTS (sorry, no make command for this currently). You can see the bits of commands here: https://github.com/black-parrot/black-parrot/blob/2e0013f3e07943c479a39bba12ba25b004311042/bp_top/test/tb/bp_tethered/Makefile.vcs#L85

Let me know how this goes! We should really document this alternative flow

flaviens, 2022-11-14:
Hi Dan, thank you very much for your answer!
I realized that my question was not detailed enough, my apologies: I took the the bp_axi_lite_top that I put into a custom SoC, where I upload the program in RAM programmatically already, and I already changed the boot PC to bypass the bootrom by starting to fetch at 0x8000000. Is there any possibility that, for example, some cache initiates a flushing cycle upon reset, and the core waits that the cache finishes this cycle before starting to fetch? This 512 looks suspiciously like a power of 2.
[image]. I observe that the fetch PC (fe.fetch_instr_v_li) increases from 0 until 0x820 (which is, 520 times PC+4) until it starts fetching to the address that I provided, maybe it helps.
Thank you very much for your support!

dpetrisko, 2022-11-14:
Oh, I see. Yes, you are right, the caches do a flush cycle upon reset (1 cycle per set). For the default configuration, there are 64 sets in the L1s, so it's ~64 cycles. If you're using the tinyparrot config, the caches have 512 sets: https://github.com/black-parrot/black-parrot/blob/6ce6523d6b3ab91cdb355d53c419f1b9ee1e1e81/bp_common/src/include/bp_common_aviary_pkgdef.svh#L59.

There's not a built-in way to fix this, as the flushing is a requirement for using hardened RAMs. I suggest trying to "concatenate" your test programs into a large binary, if that's possible?

dpetrisko, 2023-07-20:
Closing for staleness, let me know if there's anything else I can help with!

closed 2023-07-20 by dpetrisko