black-parrot/black-parrot #1140: The core fetches a single memory word when operating on cached memory area

flaviens, 2022-11-14:
Hi!
I'm considering bp_axi_lite_top with default configuration and cache sizes, integrated some trivial custom SoC where the program is pre-loaded in RAM.
When I use 0x80 as a boot address and locate my code there, everything behaves as expected. However, when I boot at 0x80000000 with the code located there, then the core fetches a single 64-bit word and seems to stop. I see clearly that the content of the fetched data is correct (and is a forward relative jump to some tens of addresses forward).
Do you have any hint on how to debug this issue?
[image]

Thanks!
Flavien

dpetrisko, 2022-11-14:
When you use 0x80 are you eventually jumping to memory at 0x80000000 or do you execute the entire program in that region?

flaviens, 2022-11-14:
Thanks for your response!
I relocate the code completely to 0x80 and execute the (quite short -- some pages) code entirely in this low area.

dpetrisko, 2022-11-14:
Got it. So the reason why it works at 0x80, but not at 0x8000000 is that the low region is executing in an "uncached" PMA region (<0x8000000).  This is causing the requests to the AXIL bus to be 8x64b uncached requests which seem to be working fine.

Looking at the code for https://github.com/black-parrot-hdk/black-parrot-subsystems/blob/main/axi/v/bp_me_axil_master.sv it's not obvious to me what's going on. But I would start debugging there. What we expect to happen is that the BP side of that adapter should produce 8x64b cached requests and the AXIL side should service them one at a time. Please follow-up here if you get a better idea of which side is going wrong!

Also: If you can post your example SoC setup somewhere to run (or email to me directly) I'd be happy to take a look when I get some time.

flaviens, 2022-11-15:
Hi Dan, thank you for your reponse!
[I prepared a repo for you to reproduce](https://github.com/flaviens/prepbp), as you suggested. Let me know if you have any issue reproducing my observations. In this repo, you will also be able to observe #1139.
This should be the default configuration, except for L2, but there is no L2 so it should not mattter.
Thanks!

dpetrisko, 2023-07-20:
Closing, but let me know if you're still having issues!

closed 2023-07-20 by dpetrisko