black-parrot/black-parrot #1142: Updating naming convention for parameterized RV handshake protocols

BrendenPage, 2022-11-15:
## Area
https://github.com/black-parrot/black-parrot/blob/master/bp_common/src/v/bsg_fifo_1r1w_rolly.sv#L7

## Issue
As we see from the above module link, the handshake protocol for this fifo (and several others located in basejump STL) has a parameter that determines the semantics of the handshake, selecting between rv->& and r->v.

This leaves us with a module with ambiguous semantics to the user depending on certain parameter values, and I believe that this should be emphasized in the naming of the signal.

My proposal is that we name this type of ready signal `ready_param_o` to emphasize that the convention is dependent on a parameter.

## Additional Context
A few other locations this is present (Basejump STL files)

https://github.com/bespoke-silicon-group/basejump_stl/blob/8f36b1985128707c7ff9f5e4349c30c5e804c300/bsg_dataflow/bsg_fifo_1r1w_small.v#L25

https://github.com/bespoke-silicon-group/basejump_stl/blob/8f36b1985128707c7ff9f5e4349c30c5e804c300/bsg_dataflow/bsg_fifo_1r1w_narrowed.v#L17

dpetrisko, 2022-11-15:
@taylor-bsg This is a pretty interesting proposal. I think ready_param_o is an elegant solution because it forces the instantiator to consider the compatibility

dpetrisko, 2023-12-21:
Completed!

closed 2023-12-21 by dpetrisko