black-parrot/black-parrot #1147: Potential PPA Optimization: I$ can have 32b banks

dpetrisko, 2022-12-11:
## Area
I$

## Current behavior
I$ has 64b banks, which are read and then muxed down to 32b.

## Desired behavior
deeper, thinner rams are generally more area-efficient. They can always be folded if this is untrue for a particular node

## Analysis
SRAM generation tables should be analysed for the design space

## Verification
Standard regression should be fine. Large programs that fill the I$ will verify.

taylor-bsg, 2022-12-11:
Presumably this is driven by the desire to write 64b at a time to match
d-cache?

dpetrisko, 2022-12-11:
64b was driven by the D$, but just by shared code history and reusing hardened rams.

We support multi-bank fill e.g. 128b fill on 2x64b banks, so for I$ this would just turn to e.g. 4x32b banks per fill. Writes shouldn't see a difference in energy. Reads however should see energy/timing improvements