black-parrot/black-parrot #1154: Black-parrot is failing in sv-tests

alaindargelas, 2023-02-21:
All the Black-parrot tests fail in sv-tests:

https://chipsalliance.github.io/sv-tests-results/?v=uhdmyosys+black-parrot+bp_be

The reason being the macro definition -DSYNTHESIS is used and triggers this macro:

`BSG_VIVADO_SYNTH_FAILS

Which makes those files containing that macro failing compilation, example:

external/basejump_stl/bsg_mem/bsg_mem_1rw_sync_mask_write_bit_synth.v

https://github.com/bespoke-silicon-group/basejump_stl/blob/9d07d182cd577acf66550f3f26ab54b6bbefaed0/bsg_mem/bsg_mem_1rw_sync_mask_write_bit_synth.v#L104

Those files are referenced by the script bp_top/syn/flist.vcs used as the project definition.

How to proceed, should there be an alternate project definition file or should we not use the -DSYNTHESIS macro?

alaindargelas, 2023-02-21:
@dpetrisko FYI.

dpetrisko, 2023-02-21:
Hi @alaindargelas thanks. 

Does Surelog have a builtin define to use? (-DSURELOG?) we can add it to this block. Currently, it's falling through to vivado because it's not detecting any of the other synthesis tools. 

https://github.com/bespoke-silicon-group/basejump_stl/blob/57db5f4aeb6b02d283a4c96033b13b8e0c37e2a8/bsg_misc/bsg_defines.v#L86

alaindargelas, 2023-02-21:
What does -DSYNTHESIS do? Besides triggering these syntax issues? Since Surelog can parse this file should we just stop using -DSYNTHESIS ?
There is no -DSURELOG but we could add one.

dpetrisko, 2023-02-21:
-DSYNTHESIS is defined by Synopsys DC/Cadence Genus and indicates that we're synthesizing for a target instead of simulating. We typically use it to trigger assertions that modules should be hardened instead of synthesized (clock gates, memories, etc.).

Removing -DSYNTHESIS is fine for the sv-tests case, but I would suggest adding a builtin SURELOG macro that we and others can hook into in the future

alaindargelas, 2023-02-23:
@dpetrisko Surelog defines the SURELOG macro now. Please add support in black-parrot.

closed 2023-02-23 by alaindargelas

dpetrisko, 2023-02-23:
Hi, latest black-parrot master has -DSURELOG support

alaindargelas, 2023-02-24:
Thanks, already visible in sv-tests (failing further now):
https://chipsalliance.github.io/sv-tests-results/?v=uhdmyosys+black-parrot+bp_be