black-parrot/black-parrot #1194: Fatal Issue : Disk Quota Exceeded

srinisy-22, 2023-09-11:
## Current behavior
Keeps giving me this error, when I try installation of BP-sim:

fatal: clone of  /black-parrot-sim/black-parrot/external/basejump_stl/imports/ramulator' failed
Failed to clone 'imports/ramulator' a second time, aborting
Failed to recurse into submodule path 'black-parrot/external/basejump_stl'
Failed to recurse into submodule path 'black-parrot'
Makefile:10: recipe for target 'checkout' failed
make: *** [checkout] Error 1
': Disk quota exceeded

## Expected behavior
I am trying to run a basic test to print Hello World

dpetrisko, 2023-09-11:
Hi, 

As the error says "': Disk quota exceeded", you're exceeding your allocated disk space. If you're running on a compute server (such as a university cluster), make sure you are using scratch space on a compute node, rather than a head node.

srinisy-22, 2023-09-11:
Hello,
 I will try that out and let you know,
Thanks

srinisy-22, 2023-09-11:
Hello I just wanted to know the size requirement for the sim installation.

dpetrisko, 2023-09-11:
Roughly 20GB during installation and 10GB after

srinisy-22, 2023-09-12:
The previous error is now gone. But Here is the new error:

fatal: unable to access 'https://git.yoctoproject.org/git/poky/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
fatal: clone of 'https://git.yoctoproject.org/git/poky' into submodule path '/black-parrot-sim/black-parrot-sdk/yocto/poky' failed
Failed to clone 'poky'. Retry scheduled
Failed to clone 'poky' a second time, aborting
Failed to recurse into submodule path 'black-parrot-sdk/yocto'
Failed to recurse into submodule path 'black-parrot-sdk'
Makefile:10: recipe for target 'checkout' failed
make: *** [checkout] Error 1

srinisy-22, 2023-09-12:
Because of this error the make -C black-parrot/bp_top/syn tire_kick command also throws an error. Kindly help me fix this.

dpetrisko, 2023-09-12:
> server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

Hi, your cluster is presumably managing your network settings and interfering with the CA certificates.

For a workaround, you could try:
git config --global http.sslVerify "false"
https://forum.gitlab.com/t/gitlab-runner-server-certificate-verification-failed/59450/16:

srinisy-22, 2023-09-12:
After executing this command, should I go through the entire installation again?

srinisy-22, 2023-09-12:
This worked like a charm when I ran make prep_lite. The object file were getting generated. But towards the end I faced with this error: pod2man bin/verilator verilator.1
pod2man bin/verilator_coverage verilator_coverage.1
help2man --no-info --no-discard-stderr --version-string=- bin/verilator_gantt -o verilator_gantt.1
make[4]: help2man: Command not found
Makefile:191: recipe for target 'verilator_gantt.1' failed
make[4]: *** [verilator_gantt.1] Error 127

Makefile:16: recipe for target 'tools_lite' failed
make[1]: *** [tools_lite] Error 2
Makefile:15: recipe for target 'prep_lite' failed
make: *** [prep_lite] Error 2

dpetrisko, 2023-09-12:
help2man is a dependency of verilator. I suggest using the docker images provided which contain all of the dependencies

srinisy-22, 2023-09-14:
So I gave up on the installation on my university system as there were too many dependencies. Instead, I am going ahead with the installation on the wsl-Ubuntu on my Windows system. After I had installed the Docker Desktop on that system, I went through with the Docker build commands including : make -C docker docker-image docker-run. I get the following error now.

[+] Building 2.0s (12/13)                                                                                docker:default
 => [internal] load .dockerignore                                                                                  0.0s
 => => transferring context: 2B                                                                                    0.0s
 => [internal] load build definition from Dockerfile.ubuntu                                                        0.0s
 => => transferring dockerfile: 1.78kB                                                                             0.0s
 => [internal] load metadata for docker.io/library/ubuntu:20.04                                                    1.6s
 => [ 1/10] FROM docker.io/library/ubuntu:20.04@sha256:33a5cc25d22c45900796a1aca487ad7a7cb09f09ea00b779e3b2026b4f  0.0s
 => CACHED [ 2/10] RUN apt update && apt install -y apt-utils tzdata git vim gettext-base uuid-dev default-jre     0.0s
 => CACHED [ 3/10] RUN ln -nsf /usr/bin/tclsh8.6 /usr/bin/tclsh                                                    0.0s
 => CACHED [ 4/10] RUN wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dear  0.0s
 => CACHED [ 5/10] RUN apt-add-repository "deb https://apt.kitware.com/ubuntu/ $(lsb_release -cs) main"     && ap  0.0s
 => CACHED [ 6/10] RUN apt clean                                                                                   0.0s
 => CACHED [ 7/10] RUN pip install orderedmultidict                                                                0.0s
 => CACHED [ 8/10] RUN getent group 0 || groupadd --gid 0 build                                                    0.0s
 => ERROR [ 9/10] RUN useradd --no-log-init --uid 0 --gid 0 --shell /bin/bash --create-home build                  0.4s
------
 > [ 9/10] RUN useradd --no-log-init --uid 0 --gid 0 --shell /bin/bash --create-home build:
0.416 useradd: UID 0 is not unique
------
Dockerfile.ubuntu:33
--------------------
  31 |
  32 |     # Use --no-log-init to deal with large userids creating giant log files
  33 | >>> RUN useradd --no-log-init --uid $USER_ID --gid $GROUP_ID --shell /bin/bash --create-home build
  34 |     #RUN useradd --no-log-init --uid 1000 --gid 1000 --shell /bin/bash --create-home build
  35 |     LABEL maintainer="Mike Patnode <mike@mpsharp.com>"
--------------------
ERROR: failed to solve: process "/bin/sh -c useradd --no-log-init --uid $USER_ID --gid $GROUP_ID --shell /bin/bash --create-home build" did not complete successfully: exit code: 4
make: *** [Makefile:9: docker-image] Error 1

dpetrisko, 2023-09-14:
Seems like docker has issues on WSL, but there may be workarounds: Docker Desktop fails to start with WSL2 based engine enabled · Issue #6971 · docker/for-wingithub.comIf you have sudo, you could also install the dependencies directly and avoid docker.

=> [ 1/10] FROM ***@***.***:33a5cc25d22c45900796a1aca487ad7a7cb09f09ea00b779e3b2026b4f  0.0s

srinisy-22, 2023-09-14:
I'm sorry, I didn't quite understand sir.

dpetrisko, 2023-09-17:
which part?

srinisy-22, 2023-09-18:
Did you mean I run this command: sudo apt-get install autoconf automake autotools-dev cmake curl default-jre libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev wget byacc device-tree-compiler python gtkwave uuid-dev vim-common virtualenv python-yaml and run the **make prep** command?

dpetrisko, 2023-09-18:
Yes my suggestion is to follow the getting started guide section for Ubuntu, as long as your WSL has Ubuntu installed. Note that we do not support WSL, however, so there may be small issues. Our officially supported builds are native Centos 7/8, native Ubuntu 18.04 and Dockerized versions of these systems

srinisy-22, 2023-09-18:
Do I need docker even after the running sudo apt-get install autoconf automake autotools-dev cmake curl default-jre libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev wget byacc device-tree-compiler python gtkwave uuid-dev vim-common virtualenv python-yaml command?

dpetrisko, 2023-09-18:
Docker is a way to virtualize your operating system for reproducible builds. If you are using your native OS you are not using docker. Here is more info: https://docs.docker.com/get-started/overview/

srinisy-22, 2023-09-20:
1. So I am currently using the Ubuntu 22.04 on VirtualBox and when try to open Docker Desktop I get this error:
Failed to connect to the bus: Operation not permitted (consider using --machine=<user>@.host --user to connect to bus of other user)
Transport endpoint is not connected

2. When I try to use --machine=<user>@.host --user, it throws me this error message.

See user logs and 'systemctl --user status docker-desktop.service' for details

3. Due to this my Docker Build is not successful and when run make prep_lite it throws me this error.
/home/jones6a9/black-parrot-sim/black-parrot/external/basejump_stl/imports/DRAMSim3/src/command_queue.cc: In member function ‘bool dramsim3::CommandQueue::QueueEmpty() const’:
/home/jones6a9/black-parrot-sim/black-parrot/external/basejump_stl/imports/DRAMSim3/src/command_queue.cc:118:21: warning: loop variable ‘q’ creates a copy from type ‘const std::vector<dramsim3::Command>’ [-Wrange-loop-construct]
  118 |     for (const auto q : queues_) {
      |                     ^
/home/jones6a9/black-parrot-sim/black-parrot/external/basejump_stl/imports/DRAMSim3/src/command_queue.cc:118:21: note: use reference type to prevent copying
  118 |     for (const auto q : queues_) {
      |                     ^
      |                     &
make[4]: Leaving directory '/home/jones6a9/black-parrot-sim/black-parrot/external/basejump_stl/bsg_test'
mv /home/jones6a9/black-parrot-sim/black-parrot/external/basejump_stl//bsg_test/libdramsim3.so /home/jones6a9/black-parrot-sim/black-parrot/install/lib/
make[3]: Leaving directory '/home/jones6a9/black-parrot-sim/black-parrot'
touch /home/jones6a9/black-parrot-sim/black-parrot/install/touch/dramsim3.e90a34120b43c4ab3a8ee0b5bca1cebccb2eee1c
make[2]: Leaving directory '/home/jones6a9/black-parrot-sim/black-parrot'
make[1]: Leaving directory '/home/jones6a9/black-parrot-sim/black-parrot'
make -C /home/jones6a9/black-parrot-sim/black-parrot-tools tools_lite
make[1]: Entering directory '/home/jones6a9/black-parrot-sim/black-parrot-tools'
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot-tools/install/bin
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot-tools/install/lib
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot-tools/install/include
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot-tools/install/touchfiles
make verilator
make[2]: Entering directory '/home/jones6a9/black-parrot-sim/black-parrot-tools'
rm -rf /home/jones6a9/black-parrot-sim/black-parrot-tools/install/touchfiles/verilator.*
cd /home/jones6a9/black-parrot-sim/black-parrot-tools; git submodule update --init --recursive --checkout /home/jones6a9/black-parrot-sim/black-parrot-tools/verilator
make verilator_build
make[3]: Entering directory '/home/jones6a9/black-parrot-sim/black-parrot-tools'
cd /home/jones6a9/black-parrot-sim/black-parrot-tools/verilator; \
      autoconf; \
      ./configure --prefix=/home/jones6a9/black-parrot-sim/black-parrot-tools/install
/bin/sh: 2: autoconf: not found
/bin/sh: 3: ./configure: not found
make[3]: *** [/home/jones6a9/black-parrot-sim/black-parrot-tools/Makefile.tools:49: verilator_build] Error 127
make[3]: Leaving directory '/home/jones6a9/black-parrot-sim/black-parrot-tools'
make[2]: *** [/home/jones6a9/black-parrot-sim/black-parrot-tools/Makefile.tools:69: /home/jones6a9/black-parrot-sim/black-parrot-tools/install/touchfiles/verilator.5d3a9eec7f4cd757f1928cb483d7a08cba509ada] Error 2
make[2]: Leaving directory '/home/jones6a9/black-parrot-sim/black-parrot-tools'
make[1]: *** [Makefile:16: tools_lite] Error 2
make[1]: Leaving directory '/home/jones6a9/black-parrot-sim/black-parrot-tools'
make: *** [Makefile:16: prep_lite] Error 2
root@22:/home/jones6a9/black-parrot-sim# make -C black-parrot/bp_top/syn tire_kick
make: Entering directory '/home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn'
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/touchfiles/verilator
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/results/verilator
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/logs/verilator
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/reports/verilator
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.sim.bp-tests.hello_world
make build.sc
make[1]: Entering directory '/home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn'
mkdir -p /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.build
cat /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/lint_settings.verilator | envsubst > /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.build/config.vlt
cd /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.build; \
      verilator --cc                  -O1 --x-assign fast --x-initial fast --threads 1 --top-module testbench -f flist.vcs config.vlt --build --exe -o simsc -Wno-timescalemod --assert --Wno-fatal --Wno-lint --Wno-style --Wno-widthconcat --Wno-unoptflat -CFLAGS -std=c++14 -CFLAGS "-I/home/jones6a9/black-parrot-sim/black-parrot/../black-parrot-tools/dromajo/include" -CFLAGS "-I/home/jones6a9/black-parrot-sim/black-parrot/install/include -I/home/jones6a9/black-parrot-sim/black-parrot/external/basejump_stl/bsg_test/" -LDFLAGS "-lpthread" -LDFLAGS "-L/home/jones6a9/black-parrot-sim/black-parrot/../black-parrot-to [... cut ...]
make: *** [/home/jones6a9/black-parrot-sim/black-parrot/bp_top/test/tb/bp_tethered/Makefile.verilator:117: /home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.sim.bp-tests.hello_world/tire_kick] Error 127
make: Leaving directory '/home/jones6a9/black-parrot-sim/black-parrot/bp_top/syn'

4. Is there a way I can get the "Hello World" .hex file and can I directly load the Verilog files in the VCS to do a basic simulation?
Kindly help me resolve these issues.

dpetrisko, 2023-09-20:
There is no reason to use docker on the vm — Just install the dependencies and follow the guide

Failed to connect to the bus: Operation not permitted (consider using ***@***.*** --user to connect to bus of other user)

When I try to use ***@***.*** --user, it throws me this error message.

srinisy-22, 2023-09-20:
I am following the guide. It fails to work.

srinisy-22, 2023-09-20:
I did install dependencies, but it keeps giving the same errors. Is there something I am missing?

srinisy-22, 2023-09-20:
When I run make -C docker docker-image docker-run, it throws an error,
 => ERROR [ 9/10] RUN useradd --no-log-init --uid 0 --gid 0 --shell /bin/bash --create-home build                                            1.0s
------                                                                                                                                            
 > [ 9/10] RUN useradd --no-log-init --uid 0 --gid 0 --shell /bin/bash --create-home build:
0.856 useradd: UID 0 is not unique
------
Dockerfile.ubuntu:33
--------------------
  31 |     
  32 |     # Use --no-log-init to deal with large userids creating giant log files
  33 | >>> RUN useradd --no-log-init --uid $USER_ID --gid $GROUP_ID --shell /bin/bash --create-home build
  34 |     
  35 |     LABEL maintainer="Mike Patnode <mike@mpsharp.com>"
--------------------
ERROR: failed to solve: process "/bin/sh -c useradd --no-log-init --uid $USER_ID --gid $GROUP_ID --shell /bin/bash --create-home build" did not complete successfully: exit code: 4
make: *** [Makefile:9: docker-image] Error 1

dpetrisko, 2023-09-20:
The error you are getting is 

/bin/sh: 2: autoconf: not found

autoconf is the first dependency listed. Therefore, I don't believe you have successfully followed the guide

srinisy-22, 2023-09-20:
I tried it in two different VMs. Let me share the results of the first VM

srinisy-22, 2023-09-20:
Here is the error that my first VM displays while running make prep_lite:
make[6]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools/verilator/src/obj_opt'
make[5]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools/verilator/src'
help2man --no-info --no-discard-stderr --version-string=- bin/verilator_gantt -o verilator_gantt.1
make[4]: help2man: Command not found
make[4]: *** [Makefile:191: verilator_gantt.1] Error 127
make[4]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools/verilator'
make[3]: *** [/home/srinisy/black-parrot-sim/black-parrot-tools/Makefile.tools:50: verilator_build] Error 2
make[3]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools'
make[2]: *** [/home/srinisy/black-parrot-sim/black-parrot-tools/Makefile.tools:69: /home/srinisy/black-parrot-sim/black-parrot-tools/install/touchfiles/verilator.5d3a9eec7f4cd757f1928cb483d7a08cba509ada] Error 2
make[2]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools'
make[1]: *** [Makefile:16: tools_lite] Error 2
make[1]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools'
make: *** [Makefile:16: prep_lite] Error 2

srinisy-22, 2023-09-20:
The first VM says the error is help2man which is a docker error as suggested by you.  But I am unable to Open Docker Desktop. Btw the VM 1 is Ubuntu 20.04

srinisy-22, 2023-09-20:
These are the docker commands I have used:
root@VM20:/home/srinisy/black-parrot-sim# systemctl --user enable docker-desktop
Failed to connect to bus: Operation not permitted
root@VM20:/home/srinisy/black-parrot-sim# docker compose version
Docker Compose version v2.21.0-desktop.1
root@VM20:/home/srinisy/black-parrot-sim# docker --version
Docker version 24.0.6, build ed223bc
root@VM20:/home/srinisy/black-parrot-sim# docker version
Client: Docker Engine - Community
 Cloud integration: v1.0.35+desktop.4
 Version:           24.0.6
 API version:       1.43
 Go version:        go1.20.7
 Git commit:        ed223bc
 Built:             Mon Sep  4 12:32:12 2023
 OS/Arch:           linux/amd64
 Context:           default

Server: Docker Engine - Community
 Engine:
  Version:          24.0.6
  API version:      1.43 (minimum version 1.12)
  Go version:       go1.20.7
  Git commit:       1a79695
  Built:            Mon Sep  4 12:32:12 2023
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.6.24
  GitCommit:        61f9fd88f79f081d64d6fa3bb1a0dc71ec870523
 runc:
  Version:          1.1.9
  GitCommit:        v1.1.9-0-gccaecfc
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0

dpetrisko, 2023-09-20:
Missing help2man is not a docker issue. It looks like Verilator has updated its dependency list recently. 

sudo apt install help2man

srinisy-22, 2023-09-20:
I have installed help2man and running make prep_lite right now.

srinisy-22, 2023-09-20:
This worked and I am able to print "Hello World". But my make prep failed. Also I dont have a VCS simulator on my VM. Is there a .hex code I can extract which represents Hello World and try it out on my University server with the same database or .hex file? (As this has vcs). Can I manually generate similar RAM input files (like .hex) to the SoC for my own testing?

dpetrisko, 2023-09-20:
You can run with Verilator as your simulator if you do not have VCS. 

The cross compiler riscv64-unknown-elf-dramfs-gcc is generated as part of prep_lite and is not dependent on any particular simualator

dpetrisko, 2023-09-20:
Oh you're asking about the other way. The simulation flow depends on the cross compiler to generate intermediate products. 

It's possible that you could run a Verilator simulation on your VM, copy those output products to your university machine, and then rerun the VCS simulation there. 

Another route would be to copy over the black-parrot-sdk/install folder. If the OS are similar enough, it may run fine and avoid needing reinstallation

srinisy-22, 2023-09-20:
I  understand, but is there a .hex file I can refer to for the Hello World Program, which servers as an input to the SoC for simulation?

srinisy-22, 2023-09-20:
I wanted to obtain several RAM files and then give them as an input to the SoC to demonstrate good functional coverage

srinisy-22, 2023-09-20:
Can I run the simulation with several of these RAM files?

dpetrisko, 2023-09-20:
> It's possible that you could run a Verilator simulation on your VM, copy those output products to your university machine, and then rerun the VCS simulation there.

> Another route would be to copy over the black-parrot-sdk/install folder. If the OS are similar enough, it may run fine and avoid needing reinstallation

These are my two suggestions to run simulations on the university machine. Either may work with some minor massaging

srinisy-22, 2023-09-20:
But the SoC requires an input RAM file for simulation right? I just wanted to know what/where is that input RAM file.

srinisy-22, 2023-09-20:
For example, the Hello World program is simulated on the SoC to print Hello World. What is the input file of the SoC for this program?

srinisy-22, 2023-09-21:
Hello Sir, I just wanted to know what is the name of the Input file (or RAM file ) for the Hello World program.

dpetrisko, 2023-09-21:
hello_world.riscv

srinisy-22, 2023-09-21:
Is this the input RAM file? It looks like a core dump after the compilation. The input is a C code or a .HEX code (Assembly code)?

dpetrisko, 2023-09-22:
It's an elf executable.

srinisy-22, 2023-09-22:
Kindly let me know where is it available in the directory. Thanks

srinisy-22, 2023-09-22:
It is available in this directory if I am not wrong black-parrot/bp_top/syn/results/verilator/bp_tethered.e_bp_default_cfg.none.sim.bp-tests.hello_world/. If this is it, the elf file is also in Core dump format. But I found a .mem file which has memory information. Is that the input to the hello world program? If so, does it mean that all my RAM files should be of this format to pursue simulation? Thanks

srinisy-22, 2023-09-22:
Basically to simulate more tests we would use the command: make -C black-parrot/bp_top/syn build.sc sim.sc COSIM_P=1. This effectively simulates "Hello World" and prints it on my console. But what if I want to simulate a different RAM file? where do I find it
?

srinisy-22, 2023-09-22:
Also what is the name of the testbench file/top level file of the SoC? Thanks

dpetrisko, 2023-09-22:
This is described in the evaluation guide. https://github.com/black-parrot/black-parrot/blob/master/docs/eval_guide.md

Programs are in black-parrot-sdk/prog. 

Different progs are selected with SUITE= PROG=. 

The testbench is described in the testbench guide https://github.com/black-parrot/black-parrot/blob/master/docs/testbench_guide.md

srinisy-22, 2023-09-22:
When I run make tools in Evaluation guide I encounter this error:
In file included from kernel/driver.cc:20:
./kernel/yosys.h:81:12: fatal error: tcl.h: No such file or directory
   81 | #  include <tcl.h>
      |            ^~~~~~~
compilation terminated.
make[3]: *** [Makefile:754: kernel/driver.o] Error 1
make[3]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools/yosys'
make[2]: *** [/home/srinisy/black-parrot-sim/black-parrot-tools/Makefile.tools:57: yosys_build] Error 2
make[2]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools'
make[1]: *** [/home/srinisy/black-parrot-sim/black-parrot-tools/Makefile.tools:70: /home/srinisy/black-parrot-sim/black-parrot-tools/install/touchfiles/yosys.9c5a60eb20104f7c320e263631c1371af9576911] Error 2
make[1]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools'
make: *** [Makefile:22: tools] Error 2

srinisy-22, 2023-09-22:
But there was an error when I ran make pull_sdk, I got this error
                                 Dload  Upload   Total   Spent    Left  Speed
100     9  100     9    0     0     48      0 --:--:-- --:--:-- --:--:--    48

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
make: *** [Makefile:30: pull_sdk] Error 2

dpetrisko, 2023-09-22:
>When I run make tools in Evaluation guide I encounter this error:
In file included from kernel/driver.cc:20:
./kernel/yosys.h:81:12: fatal error: tcl.h: No such file or directory
81 | # include <tcl.h>
| ^~~~~~~

Looks like yosys has changed its dependencies recently. I've updated the getting started guide for Ubuntu at: https://github.com/black-parrot/black-parrot-sim/blob/master/README.md. If you follow the extra dependencies and steps it should compile yosys successfully

srinisy-22, 2023-09-22:
This is the result from the run of the Update Cmake:
cmake is already the newest version (3.27.6-0kitware1ubuntu20.04.1).
The following packages were automatically installed and are no longer required:
  libjsoncpp1 librhash0
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
When I run the make tools command it gives me the same error
ake -C /home/srinisy/black-parrot-sim/black-parrot-tools/yosys
make[3]: Entering directory '/home/srinisy/black-parrot-sim/black-parrot-tools/yosys'
bash: tclsh: command not found
bash: tclsh: command not found
[  0%] Building kernel/version_9c5a60eb2.cc
[  0%] Building kernel/version_9c5a60eb2.o
[  0%] Building kernel/driver.o
In file included from kernel/driver.cc:20:
./kernel/yosys.h:81:12: fatal error: tcl.h: No such file or directory
   81 | #  include <tcl.h>
      |            ^~~~~~~
compilation terminated.
make[3]: *** [Makefile:754: kernel/driver.o] Error 1
make[3]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools/yosys'
make[2]: *** [/home/srinisy/black-parrot-sim/black-parrot-tools/Makefile.tools:57: yosys_build] Error 2
make[2]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools'
make[1]: *** [/home/srinisy/black-parrot-sim/black-parrot-tools/Makefile.tools:70: /home/srinisy/black-parrot-sim/black-parrot-tools/install/touchfiles/yosys.9c5a60eb20104f7c320e263631c1371af9576911] Error 2
make[1]: Leaving directory '/home/srinisy/black-parrot-sim/black-parrot-tools'
make: *** [Makefile:22: tools] Error 2

srinisy-22, 2023-09-26:
Hello, I was just wondering how to resolve this issue?

srinisy-22, 2023-09-26:
When I run make pull_sdk it throws me this error. 
gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
make: *** [Makefile:30: pull_sdk] Error 2

dpetrisko, 2023-09-26:
>./kernel/yosys.h:81:12: fatal error: tcl.h: No such file or directory
81 | # include <tcl.h>

should be fixed by the addition to the guide

```
sudo apt-get install apt-utils tzdata git vim gettext-base uuid-dev default-jre bash autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev wget byacc device-tree-compiler python gtkwave python-yaml pkg-config swig python3-dev pip virtualenv help2man tcl8.6-dev libreadline-dev libffi-dev software-properties-common lsb-release
# Some ubuntu distros do not provide a default symlink here
sudo ln -nsf /usr/bin/tclsh8.6 /usr/bin/tclsh
```

dpetrisko, 2023-09-26:
I do not know what is wrong with your environment for pull_sdk, you will have to investigate the error further. I just confirmed that make pull_sdk works on both Ubuntu 20.04 and Centos 7

srinisy-22, 2023-09-26:
I am using Ubuntu 20.04 and it did mention in the document that it works for that environment

dpetrisko, 2023-09-26:
Just tested Ubuntu 20.04 docker, works.

dpetrisko, 2023-09-26:
So the target is fine. There must be something wrong with your installation setup

srinisy-22, 2023-09-26:
Can you guide me more on this?

[8 more replies]