black-parrot/black-parrot #1197: Simulation behavior error

Jeremy-Jia, 2023-10-12:
When using more than one core simulation, the console output is abnormal and the core IPC is very low.[image]

dpetrisko, 2023-10-12:
Hi, that is a single threaded program that you are running with no synchronization on multiple cores. Therefore printing in an arbitrary order is expected. The "mc" programs in bp-tests are canonical examples of multithreaded programs that you can play around with. 

As for IPC, the program is very short. Core 0 finishes quickly and the waits for the other cores to finish. Cycles increases but instructions do not, so IPC drops. 

The FAIL is concerning though, seems to be a Verilator issue as VCS successfully runs this. Can you share which version of Verilator and which commit hash of BP you are using?

Jeremy-Jia, 2023-10-13:
Hi，thank you for your answer.
1. Verilator version： 4.102 2020-10-15， BP  version ： (HEAD detached at 614370f5)
2. I didn't find a suitable test case. Does "mc" need to run on Linux?
[image]
3. Is there any corresponding bare metal program that can test multi-core running performance?

dpetrisko, 2023-10-13:
1. I'd suggest using Verilator 5+ which has support for #delays and is less buggy in general. We have the recommended tools collected at https://github.com/black-parrot/black-parrot-tools
2. Sorry, mc is just the prefix, they are found here: https://github.com/black-parrot-sdk/bp-tests/
3. mc_lrsc_add, mc_rand_walk, mc_sanity and mc_work_share_sort

Jeremy-Jia, 2023-10-13:
Thanks a lot, i'll try it.
And,  can BP be configured with more than 16 cores, what should i do?

dpetrisko, 2023-10-13:
We don't test with more than 16, but there's nothing preventing it. You could copy the bp_multicore_16_cfg configuration in bp_common_aviary_pkgdef.svh and modify it to support 32 cores instead. 

However, I caution that with Verilator this with be a very slow compilation / runtime. If you're interested in studying the scalability of the coherence protocol, I would recommend looking at the testbench in bp_me/ as an example of sending arbitrary traffic into the coherence system. 

While the testbench is focused on correctness, it should be straightforward to modify it to measure performance of directed tests as well

Jeremy-Jia, 2023-10-13:
> 1. I'd suggest using Verilator 5+ which has support for #delays and is less buggy in general. We have the recommended tools collected at https://github.com/black-parrot/black-parrot-tools
> 2. Sorry, [...]
I use the recommend Verilator version(Verilator 5.012 2023-06-13 rev v5.012)，BP version :  (HEAD detached at 614370f5), but error occurs: FAIL! Core 1 stalled for 100000 cycles!, and the ipc is very low:
[image]

command is : make -C black-parrot/bp_top/syn sim.sc SUITE=bp-tests PROG=mc_amo_add CFG=e_bp_multicore_2_cfg

dpetrisko, 2023-10-21:
Looked into this a bit more -- it seems we've encountered this issue before as our CI script says:

```
# Manual, because Verilator 4.218 has scheduling issues:
#   https://github.com/verilator/verilator/issues/3278
top-mc-verilator:
  <<: *job_definition
  when: manual
  stage: test-medium
  tags:
    - verilator
  script:
    - $CI_PROJECT_DIR/ci/multicore.sh verilator $CI_CORES
 
```

Sorry about that! I don't have a good solution at the moment, as the tests pass with VCS so it is likely a Verilator error...

taylor-bsg, 2023-10-21:
So under which situations does Verilator not work?

Jeremy-Jia, 2023-10-24:
> Looked into this a bit more -- it seems we've encountered this issue before as our CI script says:
>
> ```
> # Manual, because Verilator 4.218 has scheduling issues:
> # [...]
VCS works right
[image]