black-parrot/black-parrot #1219: Significant difference in simulation time between local environment and CI Pipeline

bekbeis, 2024-04-30:
## Background information gathered
I'm trying to run the test suite for the Black Parrot using Verilator on my local machine. There is a substantial difference in execution time compared to the CI pipeline on the GitLab repository. Specifically, it takes about 1886 minutes (~31 hours) locally, whereas the CI pipeline completes it in about 80 minutes. I assume this gap could be due to the differences in simulation tools (Verilator vs. VCS) and the available hardware (32 cores on my end vs. 144 cores on the CI pipeline).

## Question
Could these factors account for such a significant difference in execution time? Could you provide any recommendations on optimizing my testing environment for faster performance (e.g., specific configurations or tool versions that might help decrease runtime on my end)?

Thank you in advance for any guidance you can provide.

dpetrisko, 2024-04-30:
Yeah VCS tends to be ~5-10x faster than Verilator for our designs (though it's very likely there's low-hanging fruit to speed it up). 

One check first, you are setting the CI scripts to use those 32 cores, right?

The best way to speed up Verilator is to feed it more RAM -- it's ravenous. Our CI machine has hundreds of gigs. The multicore designs in particular will be massively slowed down by swapping

bekbeis, 2024-05-02:
Oh, I see. Thanks a lot for your help, I appreciate it. I will try your suggestions.

closed 2024-05-02 by bekbeis

dpetrisko, 2024-05-02:
No problem, definitely contribute back if you find any Verilator tricks!