black-parrot/black-parrot #1224: Unit Testing black parrot Modules

garyz712, 2024-07-09:
Hi!
     We are trying to confirm which verilog modules are tested by each of the following black parrot tests. Is there a way to figure out their correspondence? Any insights could be helpful to us. Thank you!!!

riscv-tests (a set of unit tests for RISC-V functionality)
BEEBS (Embedded core test suite)
Coremark (Standard benchmark for core performance)
bp-tests (one-off tests which are used to test various aspects of the system)
bp-demos (demo programs showing off special features of BlackParrot firmware)
spec2000, requires a copy of the proprietary spec2000 benchmark suite
spec2006, requires a copy of the proprietary spec2006 benchmark suite
spec2017, requires a copy of the proprietary spec2017 benchmark suite
Linux (A finite test of RISC-V Linux+BusyBox boot)
DCACHE_TESTLIST
ICACHE_TESTLIST

dpetrisko, 2024-07-09:
Hi, there's not a 1:1 correspondence for the tethered example to RISC-V tests. It's a full system testbench running programs, not unit tests for specific Verilog modules

There are also specific I$ and D$ testbenches in the FE and BE respectively. These are extremely incomplete and could use love and care from outside contributors 

You can use code coverage (line/toggle) to measure completeness of verification (although this is complex and problematic in its own way). You would run subsets of the tests with coverage enabled and then merge the results to make coverage maps. This is typically a first pass of commercial verification flows

At some point we had working coverage in VCS and Verilator. I would be surprised if it still worked out of box, but if you look at the relevant makefiles it would be pretty easy to resuscitate

garyz712, 2024-07-09:
Hi!
   Thanks for your suggestion! Just to confirm, how to enable coverage and check the coverage report?
is it just adding these lines into the (BP_COMMON_DIR)/syn/Makefile.verilator?

VV_OPTS += --coverage-line
VV_OPTS += --coverage-toggle

 After running the hello world test with the coverage enabled, inside /home/build/black-parrot-sim/black-parrot/bp_top/syn/results/verilator, how to review the coverage report? I have tried to use verilator_coverage, but it says commands not found. Do I need to install verilator to enable coverage? Thank you again!

dpetrisko, 2024-07-09:
We use [basejump_stl](https://github.com/bespoke-silicon-group/basejump_stl) for the majority of BP (~85% last I checked). These modules are unit-tested. In general, the overhead of maintaining mid-block testing is greater than the utility. (leaf-block testing is very useful and integration tests are obviously good).

dpetrisko, 2024-07-09:
> >Hi!
> Thanks for your suggestion! Just to confirm, how to enable coverage and check the coverage report?
> is it just adding these lines into the (BP_COMMON_DIR)/syn/Makefile.verilator?
> 
> VV_OPTS += --coverage-line
> VV_OPTS += --coverage-toggle

Probably, although that is from ~2019 so the interface may have been updated

> After running the hello world test with the coverage enabled, inside /home/build/black-parrot-sim/black-parrot/bp_top/syn/results/verilator, how to review the coverage report? I have tried to use [...]

verilator_coverage should be installed along with verilator. If you are using black-parrot-sim, it should be in black-parrot-tools/install/bin/verilator_coverage