black-parrot/black-parrot #1229: An issue with getting started with simulation

JoshCHYeh0825, 2024-09-23:
Hello to the BlackParrot team.

As a school project, my group member and I are trying to simulate an environment for the BlackParrot processor.
After following the instructions to cloning the simulation repository (https://github.com/black-parrot/black-parrot-sim) we found out that each subsequent folder would be empty (same case with the zip file). When we recursively downloaded the submodules the repository and it also resulted in a failure. We were wondering if there is another way to download the simulation repository that does not involve manually downloading it. Thank you

dpetrisko, 2024-09-23:
BlackParrot uses git submodules: https://git-scm.com/book/en/v2/Git-Tools-Submodules

After cloning a repository with submodules, they must be initialized with:
```git submodule update --init --recursive```

This is handled in black-parrot-sim by ```make prep_lite``` in the instructions or ```make checkout``` which will do the checkout but not the build.

dpetrisko, 2024-09-23:
Note that depending on your internet connection and cpu, this may take a long time to complete